Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #1: PCF8591 pins are mirrored, no analog input circuits work

Gunnthorian, 2018-06-29:
Helping at an electronics summer camp and were using this kit to get the kids introduced to the raspberry pi. For the past week, we've been telling the kids to just skip over any circuit that requires this chip (all of the analog input circuits). offloaded the task of figuring out why the circuit wasn't working to a proper electrical engineer and he found that the PCF8591 that shipped with our kit has its pins mirrored so each and every chapter was incompletable. 

to fix this just mirror the pinouts for this chip (not rotate, mirror)

SuhaylZhao, 2018-06-30:
Thank you for choosing Freenove product.

You said that the pins of PCF8591 are mirrored.

This is not a mistake. When making circuit schematic, we always think two goals.
1. make the circuit schematic easier to understand.
2. simple and unconfused, as far as possible without crossing the line.

We usually need rotate, turn, or mirror the components on schematic. For chips with more pins, we even disorganize the order of pins and put pins with identical or similar functions together.

But on the physical connection diagram, they are consistent with the real devices. So you also see a lot of complicated interconnections.

Thanks.

zbot473, 2019-02-05:
Sorry to bump this old topic, but which way are the pins flipped?

Gunnthorian, 2019-02-06:
I don't have a kit on hand anymore but I believe they were mirrored, it was in such a way that rotating the chip didn't help. Googleing the pinouts may help @zbot473

SuhaylZhao, 2019-02-13:
Hi everyone,

Although there are no errors here, but we are still updating the view of the chip.

Please download to view the latest tutorial.

Thanks.