Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #6: Line 37, line 17 error

lxwitt, 2018-08-26:
I was working on the LightWater project and when I tried to input the python code I got the message...

"Program is starting (which is the only action I got when I did the C code, no other error, no lights, just program is starting)
Traceback (most recent call last):
   File "lightwater.py", line 37 in module
      setup ()
   file "lightwater.py", line 17 in setup
      GPIO. setup (pin GPIO out set all ledpins mode is output
Runtime error no access to dev/mem. Try running as root.

I've also had issues with the led blink with button chapter as well. The error was saying no directory or file, even though I can clearly see it in the folder manager. 

Any suggestions?

SuhaylZhao, 2018-08-28:
Hi,

We can't know your specific question. Can you send a screenshot of your question to our emailbox(support@freenove.com)?

Thanks.

closed 2018-08-28 by SuhaylZhao

FuelBuster, 2019-03-21:
Thanks gmbrill I was scratching my head on this one. seems to be an error in the code for .C maybe Python as well. Your ground switch does make it work in Python but not C.
I get the following error in C when setting it up. I am a Noob.
pi@raspberrypi:~/FreenoveKit/Code/C_Code/03.1.1_LightWater $ gcc LightWater.c -o LightWater-lwiringPi
/tmp/ccQB3kyn.o: In function `led_on':
LightWater.c:(.text+0x18): undefined reference to `digitalWrite'
/tmp/ccQB3kyn.o: In function `led_off':
LightWater.c:(.text+0x40): undefined reference to `digitalWrite'
/tmp/ccQB3kyn.o: In function `main':
LightWater.c:(.text+0x64): undefined reference to `wiringPiSetup'
LightWater.c:(.text+0xa4): undefined reference to `pinMode'
LightWater.c:(.text+0xe4): undefined reference to `delay'
LightWater.c:(.text+0x138): undefined reference to `delay'
collect2: error: ld returned 1 exit status

SuhaylZhao, 2019-03-22:
Hi,

1. Have you installed the wiringPi library?
2. Your compilation command is missing a space between "LightWater" and "-lwiringPi". It should be like this: gcc LightWater.c -o LightWater -lwiringPi

Thanks.

FuelBuster, 2019-03-22:
Thank you Suhay, it works now that I put the space between "LightWater" and "-lwiringPi". Freenove is so awesome! The projects are a great introduction to coding.

FuelBuster, 2019-03-22:
Thanks Suhayl, removing the space fixed it.
Thank you Freenove for these great projects!