Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #12: DC Motor Not Working

bkannadassan, 2019-01-20:
Hi All,

   I tried following circuit, basically I want GPIO connection without any need of relay or speed control. But apparently attached circuit never worked. Am I missing something ?, can you please help me on the same..

I see only 2 circuit with motor present.
1) With Switch
2) With Relay

I don't want either, mine should basically start with just transitior.

rgds
Balaji Kamal Kannadassan
[image]
[image]

bkannadassan, 2019-01-21:
Thanks buzzer worked, since you warned on the damage to raspberry pi. I am now testing first buzzer with external source. I see it not working, I see it working if instead of + I put in 5v and - I put in ground but not the same when I attach to + of battery and - of battery. Am I missing something ?.

-Balaji

bkannadassan, 2019-01-21:
Hi,
   Thanks I took example from Chapter 6, now its working but apparently its always making sound even when I do on/off. It stops only when I do cleanup. Attaching circuit and code for your reference. Btw When I try attaching motor though its always making sound, I see it not working. Can you let me know solution for both ?.

-Balaji

[image]
[image]
[image]

SuhaylZhao, 2019-01-21:
Hi,

1. The connection of the EC pole of the transistor is wrong. It should be exchanged.

2. The motor has the sound but does not work, may be not enough power. It is recommended that you use the power supply module. The positive (5V) is connected to the C pole of the transistor and the negative to the GND.

Please try it.

Thanks.

closed 2019-05-14 by SuhaylZhao