Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #18: led is on when executing on python

pmmonier, 2020-01-31:
On the project 02.2.1_Tablelamp, after executing "python Tablelamp.py", press the button (so the led will be on), then stop the program. If the program is executed again the led will turn on automatically when is supposed to be off, because if the button is pressed the first action is turn on the led and anything will happen because the led is already on.

SuhaylZhao, 2020-01-31:
Hi,

This program cannot record the status of the LED when the program was last closed. This program is just a demo of button controlled LED. If you need the function you said, you can save the LED state in a file, and then the next time you open the program, read the file and restore the LED state.

Thanks.

Suhayl.

pmmonier, 2020-01-31:
I think this could be solved just making sure that the led is off every time the program is stopped. You can check the commit I did, where I added on the destroy() method a conditional statement to verify if the led is on and the command "GPIO.output(ledPin,GPIO.LOW)" is executed to turn it off. I know this is just a demo, I just realized that it was happening and wanted to improve it.
Thank you.

closed 2020-01-31 by pmmonier