Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #23: Ultrasonic circuit overvolts GPIO

oboewan42, 2020-07-29:
The circuit in chapter 24 connects the Echo pin of the HC-SR04 (which runs at 5V) directly to the Pi. Since the Pi’s GPIO pins are not 5V-tolerant, this runs the risk of damaging the Pi. 

A fix is to put a 1K resistor between Echo and GPIO24, then connect two 2K resistors in series between GPIO24 and GND, creating a voltage divider to bring the input to a safe level.

SuhaylZhao, 2020-07-29:
Hi,

You are right. It is very safe to connect a 1K resistor between the gpio24 and the echo pin of the Ultrasonic module. 

Although the working voltage of raspberry pi GPIO is 3.3V, in the input mode, the GPIO is in the high resistance state, and the internal resistance is close to infinity. Therefore, in this case, the echo pin of the Ultrasonic module will not damage the GPIO of raspberry pi.

Thanks.

oboewan42, 2020-07-30:
Wrong. 

It doesn’t matter how high the internal resistance is. There is no overvoltage protection. The pins are not 5V-tolerant. The circuit, as printed, *will* damage the Pi. 

Every other Raspberry Pi resource out there - including the official documentation - explicitly warns against connecting a 5V signal to a GPIO input pin. 

Fix it.

SuhaylZhao, 2020-07-30:
Hi,

We will fix it and update soon.

Thank.

closed 2020-08-05 by SuhaylZhao