Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #25: Issue with command line

JakeJSW, 2020-09-22:
hi
I'm not sure where my issues should go but I need help with a command prompt. I am on the first tutorial labeled as blink. The issue that I'm having is when I type in cd ~/Freenove_Kit/Code/C_Code/01.1.1_Blink . The only message I get is "there is no such file or directory". I followed the above steps where I had to download the latest version of wiringpi and that all went smooth and updated. I looked through the files on the pi and have found the location of blink file. I'm not sure if its a corrupt file or if I'm doing something wrong. I've typed that line of code in like 10 times and received the same message ever time.

SuhaylZhao, 2020-09-22:
Hi,

Did you rename this folder? Please send me a screenshot.
[image]

In addition, send your questions to support@freenove.com You can also get help.

Thanks！

felix741, 2020-10-10:
python

closed 2020-11-05 by SuhaylZhao