Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #33: Java code for the starter kit ?

cmoulliard, 2021-08-24:
## Question

Do you have also the Java code for this starter kit ?

DenzelChen, 2021-08-25:
Dear customer,
Please check this folder:
Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi\Processing
Have a nice day.
Freenove

cmoulliard, 2021-08-25:
The processing - pde project is interesting but do nt allow to design the same code as you offer in C, Python. The project to be used, in this case, is : https://pi4j.com/

DenzelChen, 2021-08-25:
Sorry. Pi4j would be great. We don't have code for it yet.
Please view the websiste to try.
https://pi4j.com/getting-started/minimal-example-application/

cmoulliard, 2021-08-25:
I created the same example as C, Python in Java using Pi4j. Will share it later with a PR to you

EAGrahamJr, 2022-11-23:
Easier to work with https://github.com/mattjlewis/diozero