Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #36: no username

Louisbrown0503, 2022-08-09:


davidcomtois, 2022-08-09:
Please see: https://docs.github.com/en/repositories/creating-and-managing-repositories/cloning-a-repository

DenzelChen, 2022-08-10:
> when i put in the code "git clone --depth 1 https://github.com/freenov/Freenov_Super_Starter_Kit_for_Raspberry_Pi" and press enter its asking for a username for github ive tried my own but it doesnt work and i tried just not using a user name and pass word but it doesnt like it either need help asap

No username is needed. Please check your command again. Or you can send us a picture via support@freenove.com

Louisbrown0503, 2022-08-10:
I’ve managed to solve it it turns out I forgot a e on the end of the freenove thanks for helping anyway

closed 2022-08-10 by Louisbrown0503