Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #39: 'MbedI2C' does not name a type

djaus2, 2023-03-05:
In Acceleration detection sketch wrt MPU6050  sketch 27.1

at line 10 
```
MbedI2C iic(6,7);
```

Get compilation error **'MbedI2C' does not name a type**

Seems another library needs to be referenced.

djaus2, 2023-03-05:
Found a solution:
```
//MbedI2C iic(6,7);
TwoWire iic(i2c1,6,7);  //Use pins 6,7  
//TwoWire iic(i2c0,4,5);  //Use pins 4,5
```

TwoWire is instatiated as in wire,h
```
TwoWire(i2c_inst_t *i2c, pin_size_t sda, pin_size_t scl);
```

i2c_inst_t  is a struct   in ic2.h that  contains the following specific instance references:

In i2c.h:
``` 
typedef struct i2c_inst i2c_inst_t;

...
...

extern i2c_inst_t i2c0_inst;
extern i2c_inst_t i2c1_inst;

#define i2c0 (&i2c0_inst) ///< Identifier for I2C HW Block 0
#define i2c1 (&i2c1_inst) ///< Identifier for I2C HW Block 1
```.

Shaynee-MO, 2023-03-06:
Hi! It seemed that you are using the Raspberry Pi Pico kit, and not using the latest code.
You can download the latest one with this link:
https://github.com/Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi_Pico/archive/refs/heads/master.zip