Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #44: WiringPI outdated/no longer supported

thom2305, 2024-03-04:
Hi, I have been using your python code to code my own stuff. After all of that I was like: "It would be a great idea to try to learn C/C++". So I did that but when I installed WiringPI and ran _gpio -v_ it said :" No Hardware Line in /proc/cpuinfo"( i was removed for later kernels as it was unreliable. I thought it wasn't a big deal until I read that you need to have that line. So I started to see if you can bypass that but you can't. I was pretty disappointed. So I started searching for an alternative. I found one named pigpio Which was a good library, but there is one problem. It does not support the old WiringPI code. I did try to re-code some stuff but I was only able to code the 01.1.1 blink code to work. At this point I gave up and stayed using python. So would it be possible that you update your guides(every guide you have as they all have that same error) to use pigpio or some other lib for gpio usage in C/C++ code.
 
Thank you

Thom

Shaynee-MO, 2024-03-05:
Hi!
WiringPi now supports Raspberry Pi 5 and the latest Raspberry Pi OS.
Please update the WiringPi library and try again.
Before updating, please uninstall the original one first.

Run the following commands one by one to uninstall it and delete it thoroughtly:
cd WiringPi
./build uninstall 
sudo apt-get remove wiringpi
sudo apt-get purge wiringpi
hash -r

Download and install the latest WiringPi:
git clone https://github.com/WiringPi/WiringPi
cd WiringPi
./build

Run the gpio command to check the installation:
gpio -v

If you are using RPi 5, you need to update WiringPi to 3.0 version.

Shaynee-MO, 2024-03-05:
Hi!
Please run the following commands one by one and send us screenshots of the results.
getconf LONG_BIT
cat /proc/cpuinfo
cat /etc/rpi-issue

Run the gpio to check the installation of wiringpi library.
gpio -v 

If the version is lower than 3.0, please update it first.

Before updating, please uninstall the original one first.

Run the following commands one by one to uninstall it and delete it thoroughly:
cd WiringPi
./build uninstall
sudo apt-get remove wiringpi
sudo apt-get purge wiringpi
hash -r

Download and install the latest WiringPi:
git clone https://github.com/WiringPi/WiringPi
cd WiringPi
./build

Run the gpio command to check the installation:
gpio -v

thom2305, 2024-03-11:
Since Wiring Pi Was recently updated I wasnt able to know that it had an update, now everything works.

closed 2024-03-11 by thom2305