Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi #46: Raspberry Pi 4 - Processing Issue

Jess338, 2024-09-06:
My raspberry pi for some reason wont execute the "processing" file you need to install to start the kit, do you know of any work arounds or ways to get it to work. To be more exact I can click execute but it loads for a bit and doesn't actually open anything.

Shaynee-MO, 2024-09-07:
Hi! We have updated the Processing tutorial. Please download the latest one and try again.

Here are videos for chapter 0 and chapter 1 for your reference.
https://youtu.be/atfssDk2urw
https://youtu.be/N2diyg_Anc0

Jess338, 2024-09-09:
> <Freenove/Freenove_Ultimate_Starter_Kit_for_Raspberry_Pi/issues/46/2335024986
> @github.com>

Thank you

Akspower, 2025-03-23:
My raspberry pi for some reason wont execute the "processing" file you need to install to start the kit, do you know of any work arounds or ways to get it to work. To be more exact I can click execute but it loads for a bit and doesn't actually open anything.

Shaynee-MO, 2025-03-24:
@Akspower 
To help us troubleshoot, please provide the following information.
1. Run the following commands one by one and send us screenshots of the results.
getconf LONG_BIT
cat /proc/cpuinfo
cat /etc/rpi-issue
2. Tell us the version of Processing you have installed.
3. Send us a video to show the issue.
We will check and offer solutions. Many thanks for your help.